JunyuanDeng/NeRF-LOAM #5: Sharing CUDA tensors

Zhangjyhhh, 2023-10-11:
hi
Thanks for your excellent work. when I run the code " python demo/run.py configs/kitti/kitti_00.yaml" I met  some issue ,i set "end frame =20" in the /config/kitti/kitti_00.yaml. my pytorch version is 1.10. cuda versio  is 11.1. could you tell me how to solve this issue?

`PatchWorkpp::PatchWorkpp() - INITIALIZATION COMPLETE
Decoder(
  (pe): Same()
  (pts_linears): ModuleList(
    (0): Linear(in_features=16, out_features=256, bias=True)
    (1): Linear(in_features=256, out_features=256, bias=True)
  )
  (sdf_out): Linear(in_features=256, out_features=1, bias=True)
)
******* initializing first_frame: 0
initializing the first frame ...
mapping process started!!!!!!!!!
frame id 1
trans  tensor([0., 0., 0.], device='cuda:0')
insert keyframe
PatchWorkpp::PatchWorkpp() - INITIALIZATION COMPLETE
******* tracking process started! *******
tracking frame:   5%|█▍                          | 1/20 [00:10<03:18, 10.47s/it]/home/jyzhang/anaconda3/envs/nerfloam/lib/python3.8/site-packages/torch/functional.py:445: UserWarning: torch.meshgrid: in an upcoming release, it will be required to pass the indexing argument. (Triggered internally at  /opt/conda/conda-bld/pytorch_1634272128894/work/aten/src/ATen/native/TensorShape.cpp:2157.)
  return _VF.meshgrid(tensors, **kwargs)  # type: ignore[attr-defined]
(1, 4, 4)
tracking frame:  10%|██▊                         | 2/20 [00:12<01:40,  5.59s/it]frame id 2
trans  tensor([0.6877, 0.0027, 0.0078], device='cuda:0', grad_fn=<SubBackward0>)
tracking frame:  15%|████▏                       | 3/20 [00:14<01:05,  3.86s/it]frame id 3
trans  tensor([1.3820, 0.0112, 0.0105], device='cuda:0', grad_fn=<SubBackward0>)
tracking frame:  20%|█████▌                      | 4/20 [00:16<00:49,  3.09s/it]frame id 4
trans  tensor([2.1094, 0.0249, 0.0096], device='cuda:0', grad_fn=<SubBackward0>)
tracking frame:  25%|███████                     | 5/20 [00:18<00:40,  2.73s/it]frame id 5
trans  tensor([2.8489, 0.0413, 0.0128], device='cuda:0', grad_fn=<SubBackward0>)
tracking frame:  30%|████████▍                   | 6/20 [00:20<00:35,  2.53s/it]frame id 6
trans  tensor([3.5876, 0.0570, 0.0211], device='cuda:0', grad_fn=<SubBackward0>)
tracking frame:  35%|█████████▊                  | 7/20 [00:22<00:31,  2.40s/it]frame id 7
trans  tensor([4.3582, 0.0895, 0.0347], device='cuda:0', grad_fn=<SubBackward0>)
tracking frame:  40%|███████████▏                | 8/20 [00:24<00:28,  2.35s/it]frame id 8
trans  tensor([5.1436, 0.1161, 0.0417], device='cuda:0', grad_fn=<SubBackward0>)
tracking frame:  45%|████████████▌               | 9/20 [00:27<00:25,  2.34s/it]frame id 9
trans  tensor([5.9294, 0.1443, 0.0319], device='cuda:0', grad_fn=<SubBackward0>)
tracking frame:  50%|█████████████▌             | 10/20 [00:29<00:23,  2.35s/it]frame id 10
trans  tensor([6.7445, 0.1803, 0.0405], device='cuda:0', grad_fn=<SubBackward0>)
tracking frame:  55%|██████████████▊            | 11/20 [00:31<00:20,  2.33s/it]frame id 11
trans  tensor([7.5590, 0.2162, 0.0498], device='cuda:0', grad_fn=<SubBackward0>)
tracking frame:  60%|████████████████▏          | 12/20 [00:34<00:19,  2.38s/it]frame id 12
trans  tensor([8.3817, 0.2561, 0.0583], device='cuda:0', grad_fn=<SubBackward0>)
insert keyframe
********** current num kfs: 2 **********
tracking frame:  65%|█████████████████▌         | 13/20 [00:36<00:16,  2.39s/it]frame id 13
trans  tensor([9.2186, 0.2980, 0.0541], device='cuda:0', grad_fn=<SubBackward0>)
tracking frame:  70%|██████████████████▉        | 14/20 [00:39<00:14,  2.41s/it]frame id 14
trans  tensor([10.0634,  0.3389,  0.0636], device='cuda:0', grad_fn=<SubBackward0>)
tracking frame:  75%|████████████████████▎      | 15/20 [00:41<00:12,  2.42s/it]frame id 15
trans  tensor([10.9102,  0.3870,  0.0693], device='cuda:0', grad_fn=<SubBackward0>)
tracking frame:  80%|█████████████████████▌     | 16/20 [00:44<00:09,  2.47s/it]frame id 16
trans  tensor([11.7786,  0.4310,  0.0806], device='cuda:0', grad_fn=<SubBackward0>)
tracking frame:  85%|██████████████████████▉    | 17/20 [00:46<00:07,  2.53s/it]frame id 17
trans  tensor([12.6559,  0.4845,  0.0818], device='cuda:0', grad_fn=<SubBackward0>)
tracking frame:  90%|████████████████████████▎  | 18/20 [00:49<00:05,  2.57s/it]frame id 18
trans  tensor([13.5310,  0.5446,  0.0908], device='cuda:0', grad_fn=<SubBackward0>)
tracking frame:  95%|█████████████████████████▋ | 19/20 [00:52<00:02,  2.58s/it]frame id 19
trans  tensor([14.4132,  0.6029,  0.0900], device='cuda:0', grad_fn=<SubBackward0>)
tracking frame: 100%|███████████████████████████| 20/20 [00:54<00:00,  2.75s/it]
========== stop_mapping set ==========
******* tracking process died *******
frame id 20
trans  tensor([15.3066,  0.6636,  0.1000], device='cuda:0', grad_fn=<SubBackward0>)
frame id 21
trans  tensor([16.2136,  0.7190,  0.1097], device='cuda:0', grad_fn=<SubBackward0>)
******* extracting mesh without replay *******
********** post-processing steps **********
 post-processing steps: 100%|█████████████████████| 3/3 [00:06<00:00,  2.17s/it]
******* extracting final mesh *******
(21, 4, 4)
******* mapping process died *******
[W CudaIPCTypes.cpp:15] Producer process has been terminated before all shared CUDA tensors released. See Note [Sharing CUDA tensors]
[W CudaIPCTypes.cpp:15] Producer process has been terminated before all shared CUDA tensors released. See Note [Sharing CUDA tensors]
/home/jyzhang/anaconda3/envs/nerfloam/lib/python3.8/multiprocessing/resource_tracker.py:216: UserWarning: resource_tracker: There appear to be 3 leaked semaphore objects to clean up at shutdown
  warnings.warn('resource_tracker: There appear to be %d '
`

JunyuanDeng, 2023-10-11:
Hi,

This is just a warning and does not affect the result. I also have no idea how to solve this warning.

Zhangjyhhh, 2023-10-11:
thanks for your reply I find the final mesh result

closed 2023-10-12 by Zhangjyhhh