JunyuanDeng/NeRF-LOAM #9: The issue of CUDA out of memory when running on the sequence00 of maicity

laliwang, 2023-12-06:
The OOM error happens when I run the command: `python demo/run.py configs/maicity/maicity_00.yaml`. Since I have ran the code on the GPU of RTX 4090 with 24GB memory available, the OOM shouldn't happen as you mentioned in your README. And which confused me greatly is that there is still memory avaliable when OOM happens. The outputs are shown as follows:

`RuntimeError: CUDA out of memory. Tried to allocate 5.54 GiB (GPU 0; 23.65 GiB total capacity; 1.91 GiB already allocated; 5.50 GiB free; 1.92 GiB reserved in total by PyTorch) If reserved memory is >> allocated memory try setting max_split_size_mb to avoid fragmentation.  See documentation for Memory Management and PYTORCH_CUDA_ALLOC_CONF`

And it is worth noting that the OOM tends to happen in the process of 246/699 in sequence00 of maicity.

JunyuanDeng, 2023-12-07:
Normally, this program have tested all on 24GB GPU, it may not have problems.

I notice that you have 24GB total capacity and 1 GB allocated, but only 5.5GB free, maybe you have other programs that are running?

closed 2023-12-10 by laliwang

woojinKwon, 2024-07-10:
Hello, I am also using an RTX 4090, and when I run run.py, I encounter an error message during the mapping process saying, "Encounter a bug while Tracking, currently not be fixed, Restarting!!". Could you please share your nvcc version and torch version? If you know any additional solutions to this problem, please let me know.

JunyuanDeng, 2024-07-10:
Possiblely, it is due to the pose error causing no intersection of ray and prebuilt map.

woojinKwon, 2024-07-10:
Thank you for your response. However, I am encountering this issue while using the provided maicity_00 dataset and maicity_00.yaml file, so it seems to be a version compatibility problem.

JunyuanDeng, 2024-07-10:
They are all listed [here](https://github.com/JunyuanDeng/NeRF-LOAM?tab=readme-ov-file#environment-setup).