JunyuanDeng/NeRF-LOAM #13: patchwork_module_path ="/xxx/patchwork-plusplus/build/python_wrapper"

Abin-star, 2023-12-18:
请问我编译后build下找不到python_wrapper

JunyuanDeng, 2023-12-19:
一般来说，编译后是可以找到的。可能是你编译过程出错了，如果您能提供更多编译时的信息或者其他信息，我能更清楚定位到问题。

Abin-star, 2023-12-19:
(base) abin@abin:~/patchwork-plusplus/build$ cmake ..
-- The C compiler identification is GNU 9.4.0
-- The CXX compiler identification is GNU 9.4.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building examples for c++
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Failed
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Found OpenMP_C: -fopenmp (found version "4.5") 
-- Found OpenMP_CXX: -fopenmp (found version "4.5") 
-- Found OpenMP: TRUE (found version "4.5")  
-- Found X11: /usr/include   
-- Looking for XOpenDisplay in /usr/lib/x86_64-linux-gnu/libX11.so;/usr/lib/x86_64-linux-gnu/libXext.so
-- Looking for XOpenDisplay in /usr/lib/x86_64-linux-gnu/libX11.so;/usr/lib/x86_64-linux-gnu/libXext.so - found
-- Looking for gethostbyname
-- Looking for gethostbyname - found
-- Looking for connect
-- Looking for connect - found
-- Looking for remove
-- Looking for remove - found
-- Looking for shmat
-- Looking for shmat - found
-- Looking for IceConnectionNumber in ICE
-- Looking for IceConnectionNumber in ICE - found
-- Found OpenGL: /usr/lib/x86_64-linux-gnu/libOpenGL.so   
-- Found Open3D: /usr/local/lib/cmake/Open3D/Open3DConfig.cmake (found version "0.15.1") 
-- Found Open3D 0.15.1
-- Configuring done
-- Generating done
-- Build files have been written to: /home/abin/patchwork-plusplus/build

(base) abin@abin:~/patchwork-plusplus/build$ make
Scanning dependencies of target patchworkpp
[ 16%] Building CXX object patchworkpp/CMakeFiles/patchworkpp.dir/src/patchworkpp.cpp.o
[ 33%] Linking CXX static library libpatchworkpp.a
[ 33%] Built target patchworkpp
Scanning dependencies of target demo_sequential
[ 50%] Building CXX object CMakeFiles/demo_sequential.dir/examples/cpp/demo_sequential.cpp.o
[ 66%] Linking CXX executable examples/cpp/demo_sequential
[ 66%] Built target demo_sequential
Scanning dependencies of target demo_visualize
[ 83%] Building CXX object CMakeFiles/demo_visualize.dir/examples/cpp/demo_visualize.cpp.o
[100%] Linking CXX executable examples/cpp/demo_visualize
[100%] Built target demo_visualize

JunyuanDeng, 2023-12-19:
看上去已经装好了，你有运行过`pip install .` 吗？

Abin-star, 2023-12-19:
运行了，我分别尝试了先c++编译再python编译和先python编译再c++编译，结果都是没有在build下生成python_wrapper，编译过程也没有错误，您能和我讲解一下这个文件的作用嘛

JunyuanDeng, 2023-12-19:
主要是因为原来是用c++写的，用python调用的话，就需要这么一个接口，里面有一个.so文件。如果您能运行`pip install .`我猜测他已经装在了你系统里面了，你可以看看能不能运行他里面的demo，python版本的。如果可以的话，那就不需要patchwork_module_path 了

bason-ren, 2023-12-19:
恩恩，是这样的，谢谢您啦

closed 2023-12-19 by JunyuanDeng