JunyuanDeng/NeRF-LOAM #14: python demo/run.py configs/kitti/kitti_00.yaml

bason-ren, 2023-12-20:
您好，我在运行run.py时遇到如下报错，请问您知道具体原因麻

(torch) abin@abin:~/NeRF-LOAM$ python demo/run.py configs/kitti/kitti_00.yaml 
PatchWorkpp::PatchWorkpp() - INITIALIZATION COMPLETE
Decoder(
  (pe): Same()
  (pts_linears): ModuleList(
    (0): Linear(in_features=16, out_features=256, bias=True)
    (1): Linear(in_features=256, out_features=256, bias=True)
  )
  (sdf_out): Linear(in_features=256, out_features=1, bias=True)
)
******* initializing first_frame: 0
Bus error (core dumped)

JunyuanDeng, 2023-12-20:
这个问题我们确实没遇到，您能具体定位到是哪个地方出错吗？

bason-ren, 2023-12-23:
已解决，谢谢您

closed 2023-12-27 by JunyuanDeng

ArthurHoumo, 2024-05-21:
> 已解决，谢谢您

你好，遇到了相同问题，请问是怎么解决的呢？

boyang9602, 2024-08-26:
@JunyuanDeng @ArthurHoumo 
我也遇到同样的问题了，我是在docker里，用的是`nvidia/cuda:11.1.1-cudnn8-devel-ubuntu20.04` base，里面装了python3.8, torch 1.10

问题可能是出在multiprocessing这里，我把 
https://github.com/JunyuanDeng/NeRF-LOAM/blob/2fe4e8d8dd9a8da12624ba8e8e4f49d851ce8c92/src/nerfloam.py#L23
注释掉就没有Bus error了，但是进度条卡在0% 不动，应该还有别的问题