JunyuanDeng/NeRF-LOAM #16: AttributeError: module 'grid' has no attribute 'svo_intersect'

DeepDuke, 2023-12-27:
```

PatchWorkpp::PatchWorkpp() - INITIALIZATION COMPLETE
Decoder(
  (pe): Same()
  (pts_linears): ModuleList(
    (0): Linear(in_features=16, out_features=256, bias=True)
    (1): Linear(in_features=256, out_features=256, bias=True)
  )
  (sdf_out): Linear(in_features=256, out_features=1, bias=True)
)
******* initializing first_frame: 0
initializing the first frame ...
mapping process started!!!!!!!!!
frame id 1
trans  tensor([0., 0., 0.], device='cuda:0')
insert keyframe
Process Process-2:
Traceback (most recent call last):
  File "/home/spacex/Data/noetic_cudagl_workspace/Study/anaconda3/envs/nerfloam/lib/python3.9/multiprocessing/process.py", line 315, in _bootstrap
    self.run()
  File "/home/spacex/Data/noetic_cudagl_workspace/Study/anaconda3/envs/nerfloam/lib/python3.9/multiprocessing/process.py", line 108, in run
    self._target(*self._args, **self._kwargs)
  File "/home/spacex/Data/noetic_cudagl_workspace/Study/NeRF-LOAM/src/mapping.py", line 110, in spin
    self.do_mapping(share_data, tracked_frame, selection_method='current')
  File "/home/spacex/Data/noetic_cudagl_workspace/Study/NeRF-LOAM/src/mapping.py", line 184, in do_mapping
    bundle_adjust_frames(
  File "/home/spacex/Data/noetic_cudagl_workspace/Study/NeRF-LOAM/src/variations/render_helpers.py", line 394, in bundle_adjust_frames
    final_outputs = render_rays(
  File "/home/spacex/Data/noetic_cudagl_workspace/Study/NeRF-LOAM/src/variations/render_helpers.py", line 211, in render_rays
    intersections, hits = ray_intersect(
  File "/home/spacex/Data/noetic_cudagl_workspace/Study/anaconda3/envs/nerfloam/lib/python3.9/site-packages/torch/autograd/grad_mode.py", line 28, in decorate_context
    return func(*args, **kwargs)
  File "/home/spacex/Data/noetic_cudagl_workspace/Study/NeRF-LOAM/src/variations/voxel_helpers.py", line 534, in ray_intersect
    pts_idx, min_depth, max_depth = svo_ray_intersect(
  File "/home/spacex/Data/noetic_cudagl_workspace/Study/NeRF-LOAM/src/variations/voxel_helpers.py", line 110, in forward
    inds, min_depth, max_depth = _ext.svo_intersect(
AttributeError: module 'grid' has no attribute 'svo_intersect'
^CTraceback (most recent call last):
  File "/home/spacex/Data/noetic_cudagl_workspace/Study/NeRF-LOAM/demo/run.py", line 26, in <module>
    slam.start()
  File "/home/spacex/Data/noetic_cudagl_workspace/Study/NeRF-LOAM/src/nerfloam.py", line 45, in start
    sleep(20)
KeyboardInterrupt
```

Hi, I met this error, what's your version for the lib `grid`?

JunyuanDeng, 2023-12-27:
Hi,

It seems that you have not installed successfully the [extra modules required](https://github.com/JunyuanDeng/NeRF-LOAM?tab=readme-ov-file#environment-setup:~:text=After%20you%20have,sh%20install.sh).

As you can see that this function, `svo_intersect` is bound in this file: `third_party/sparse_voxels/src/binding.cpp`.

DeepDuke, 2023-12-27:
Hi, it's solved now. TKS! But I met another problem when using my own dataset in `*.pcd` format. 
```
transport frame: 54.28 ms
tracking frame:   2%|██▋                                                                                                                                     | 2/99 [00:03<02:04,  1.29s/it]mapping_add_optim: 0.09 ms
mapping sample_rays: 1.68 ms
ray_intersect: 3.29 ms
Encouter a bug while Mapping, currently not be fixed, Continue!!
before track1111: 1.50 ms
before track2222: 1.14 ms
track sample_rays: 0.36 ms
ray_intersect: 1.61 ms
ray_sample: 1.87 ms
get_features: 1.21 ms
render_core: 1.07 ms
track render_rays: 12.81 ms
track loss_criteria: 1.08 ms
track backward step: 3.39 ms
Encouter a bug while Mapping, currently not be fixed, Continue!!
Encouter a bug while Mapping, currently not be fixed, Continue!!
Encouter a bug while Mapping, currently not be fixed, Continue!!
Encouter a bug while Mapping, currently not be fixed, Continue!!
Encouter a bug while Mapping, currently not be fixed, Continue!!
Encouter a bug while Mapping, currently not be fixed, Continue!!
Encouter a bug while Mapping, currently not be fixed, Continue!!
Encouter a bug while Mapping, currently not be fixed, Continue!!
Encouter a bug while Mapping, currently not be fixed, Continue!!
Encouter a bug while Mapping, currently not be fixed, Continue!!
Encouter a bug while Mapping, currently not be fixed, Continue!!
Encouter a bug while Mapping, currently not be fixed, Continue!!
Encouter a bug while Mapping, currently not be fixed, Continue!!
Encouter a bug while Mapping, currently not be fixed, Continue!!
Encouter a bug while Mapping, currently not be fixed, Continue!!
Encouter a bug while Mapping, currently not be fixed, Continue!!
Encouter a bug while Mapping, currently not be fixed, Continue!!
Encouter a bug while Mapping, currently not be fixed, Continue!!
Encouter a bug while Mapping, currently not be fixed, Continue!!
Encouter a bug while Mapping, currently not be fixed, Continue!!
Encouter a bug while Mapping, currently not be fixed, Continue!!
bundle_adjust_frames: 230.21 ms
do_mapping: 233.05 ms
frame id 2

```

DeepDuke, 2023-12-27:
DO I need to tune some params to make it work ?

DeepDuke, 2023-12-27:
It seems no hits during render... Any suggestion for tuning parameters?

DeepDuke, 2023-12-27:
Oh, I change the min_depth to be smaller. It can run now.

JunyuanDeng, 2023-12-27:
OK, glad to hear that you have solved it. It seems that you trained on personal dataset, maybe you need to change hyperparameters like learning rate for tracking or mapping.

DeepDuke, 2023-12-27:
Well, I collect a lidar sequence from Carla simulator. The scene is similar to the maicity dataset. But it seems that the hyperparameters are not easy to tune for a complete mesh reconstruction.... The mesh quality is very poor:
[image] 
I tried to tune ` learning rate, step_size, voxel_size, window_size`, but got no effect.

JunyuanDeng, 2023-12-27:
It seems that the pose got some error, is the predicted pose correct?

DeepDuke, 2023-12-27:
No. The predicted pose deviated too much from the ground truth. The average translation error is about 60 meters...

JunyuanDeng, 2023-12-27:
Ok, we find at least one problem, if the collected lidar moves fast, the learning rate should be higher, maybe 0.08 or 0.1 and vice-verse.

DeepDuke, 2023-12-27:
The car moves at roughly 20m/s. Should I increase the learning rate?

JunyuanDeng, 2023-12-28:
yes, that quite fast, maicity moves around 10m/s, the gap bewteen two frames is about 0.1second. You can try to adjust the tracking learning rate to maybe 0.08 or 0.1

closed 2024-01-13 by JunyuanDeng

qgh66, 2024-02-04:
PatchWorkpp::PatchWorkpp() - INITIALIZATION COMPLETE
Decoder(
  (pe): Same()
  (pts_linears): ModuleList(
    (0): Linear(in_features=16, out_features=256, bias=True)
    (1): Linear(in_features=256, out_features=256, bias=True)
  )
  (sdf_out): Linear(in_features=256, out_features=1, bias=True)
)
******* initializing first_frame: 0
initializing the first frame ...
mapping process started!!!!!!!!!
frame id 1
trans  tensor([0., 0., 0.], device='cuda:0')
insert keyframe
Process Process-2:
Traceback (most recent call last):
  File "/usr/lib/python3.8/multiprocessing/process.py", line 315, in _bootstrap
    self.run()
  File "/usr/lib/python3.8/multiprocessing/process.py", line 108, in run
    self._target(*self._args, **self._kwargs)
  File "/home/NeRF-LOAM/src/mapping.py", line 107, in spin
    self.do_mapping(share_data, tracked_frame, selection_method='current')
  File "/home/NeRF-LOAM/src/mapping.py", line 179, in do_mapping
    bundle_adjust_frames(
  File "/home/NeRF-LOAM/src/variations/render_helpers.py", line 394, in bundle_adjust_frames
    final_outputs = render_rays(
  File "/home/NeRF-LOAM/src/variations/render_helpers.py", line 211, in render_rays
    intersections, hits = ray_intersect(
  File "/usr/local/lib/python3.8/dist-packages/torch/autograd/grad_mode.py", line 27, in decorate_context
    return func(*args, **kwargs)
  File "/home/NeRF-LOAM/src/variations/voxel_helpers.py", line 534, in ray_intersect
    pts_idx, min_depth, max_depth = svo_ray_intersect(
  File "/home/NeRF-LOAM/src/variations/voxel_helpers.py", line 110, in forward
    inds, min_depth, max_depth = _ext.svo_intersect(
AttributeError: module 'grid' has no attribute 'svo_intersect'