JunyuanDeng/NeRF-LOAM #18: IndexError: select(): index 108605 out of range for tensor of size [69829, 4] at dimension 0

jxbb233, 2024-01-02:
你好，
由于 #17 的原因，我尝试使用subscene的branch运行demo，但是遇到了如下报错。
```
/usr/local/lib/python3.8/dist-packages/torch/functional.py:599: UserWarning: torch.meshgrid: in an upcoming release, it will be required to pass the indexing argument. (Triggered internally at  ../aten/src/ATen/native/TensorShape.cpp:2315.)
  return _VF.meshgrid(tensors, **kwargs)  # type: ignore[attr-defined]
Process Process-2:
Traceback (most recent call last):
  File "/usr/lib/python3.8/multiprocessing/process.py", line 315, in _bootstrap
    self.run()
  File "/usr/lib/python3.8/multiprocessing/process.py", line 108, in run
    self._target(*self._args, **self._kwargs)
  File "/data/NeRF-LOAM-subscene/src/mapping.py", line 113, in spin
    self.create_voxels(tracked_frame)
  File "/data/NeRF-LOAM-subscene/src/mapping.py", line 333, in create_voxels
    self.update_grid_features()
  File "/usr/local/lib/python3.8/dist-packages/torch/autograd/grad_mode.py", line 27, in decorate_context
    return func(*args, **kwargs)
  File "/data/NeRF-LOAM-subscene/src/mapping.py", line 363, in update_grid_features
    voxels, children, features = self.svo.get_centres_and_children()
IndexError: select(): index 108605 out of range for tensor of size [69829, 4] at dimension 0
```
我运行时使用的代码是
```
python demo/run.py configs/kitti/kitti_09.yaml
```
感谢您的帮助！

JunyuanDeng, 2024-01-03:
您好，我在运行这段代码的时候没有发现有什么问题。
```
PatchWorkpp::PatchWorkpp() - INITIALIZATION COMPLETE
Decoder(
  (pe): Same()
  (pts_linears): ModuleList(
    (0): Linear(in_features=16, out_features=256, bias=True)
    (1): Linear(in_features=256, out_features=256, bias=True)
  )
  (sdf_out): Linear(in_features=256, out_features=1, bias=True)
)
******* initializing first_frame: 0
initializing the first frame ...
mapping process started!!!!!!!!!
frame id 1
trans  tensor([0., 0., 0.], device='cuda:0')
insert keyframe
PatchWorkpp::PatchWorkpp() - INITIALIZATION COMPLETE
******* tracking process started! *******
tracking frame:   0%|▏                                                                                                            | 2/1590 [00:17<3:26:19,  7.80s/it]frame id 2
trans  tensor([ 0.3031, -0.0071,  0.0090], device='cuda:0', grad_fn=<SubBackward0>)
tracking frame:   0%|▏                                                                                                            | 3/1590 [00:20<2:18:54,  5.25s/it]frame id 3
trans  tensor([ 0.5925, -0.0283,  0.0078], device='cuda:0', grad_fn=<SubBackward0>)
tracking frame:   0%|▎                                                                                                            | 4/1590 [00:21<1:42:52,  3.89s/it]frame id 4
trans  tensor([ 0.8979, -0.0399,  0.0087], device='cuda:0', grad_fn=<SubBackward0>)
tracking frame:   0%|▎                                                                                                            | 5/1590 [00:24<1:31:08,  3.45s/it]frame id 5
trans  tensor([ 1.2074, -0.0474,  0.0084], device='cuda:0', grad_fn=<SubBackward0>)
tracking frame:   0%|▍                                                                                                            | 6/1590 [00:27<1:23:25,  3.16s/it]frame id 6
trans  tensor([ 1.5337, -0.0493,  0.0092], device='cuda:0', grad_fn=<SubBackward0>)
tracking frame:   0%|▍                                                                                                            | 7/1590 [00:30<1:20:11,  3.04s/it]frame id 7
trans  tensor([ 1.8604, -0.0443,  0.0176], device='cuda:0', grad_fn=<SubBackward0>)
tracking frame:   1%|▌                                                                                                            | 8/1590 [00:32<1:17:47,  2.95s/it]frame id 8
trans  tensor([ 2.1960, -0.0314,  0.0227], device='cuda:0', grad_fn=<SubBackward0>)
tracking frame:   1%|▌                                                                                                            | 9/1590 [00:35<1:15:58,  2.88s/it]frame id 9
trans  tensor([ 2.5394e+00, -2.4414e-03,  2.5757e-02], device='cuda:0',
       grad_fn=<SubBackward0>)
frame id 10
trans  tensor([2.8971, 0.0330, 0.0234], device='cuda:0', grad_fn=<SubBackward0>)
frame id 11
trans  tensor([3.2632, 0.0710, 0.0236], device='cuda:0', grad_fn=<SubBackward0>)
********** post-processing steps **********
 post-processing steps: 100%|██████████████████████████████████████████████████████████████████████████████████████████████████████████| 2/2 [00:02<00:00,  1.47s/it]
/home/evsjtu2/miniconda3/envs/djyvoxfusion/lib/python3.8/site-packages/torch/functional.py:445: UserWarning: torch.meshgrid: in an upcoming release, it will be required to pass the indexing argument. (Triggered internally at  /opt/conda/conda-bld/pytorch_1639180595841/work/aten/src/ATen/native/TensorShape.cpp:2157.)
  return _VF.meshgrid(tensors, **kwargs)  # type: ignore[attr-defined]
(11, 4, 4)
******* initializing restart_frame: 10
tracking frame:   1%|▋                                                                                                           | 10/1590 [00:50<2:56:19,  6.70s/it]frame id 11
trans  tensor([3.2632, 0.0710, 0.0236], device='cuda:0')
insert keyframe
tracking frame:   1%|▊                                                                                                           | 12/1590 [01:02<2:35:33,  5.91s/it]frame id 12
trans  tensor([3.6542, 0.1188, 0.0378], device='cuda:0', grad_fn=<SubBackward0>)
tracking frame:   1%|▉                                                                                                           | 13/1590 [01:04<2:06:12,  4.80s/it]frame id 13
trans  tensor([ 4.0375,  0.1584, -0.0381], device='cuda:0', grad_fn=<SubBackward0>)
```
看上去是在svo，也就是第三方库时候存在的问题，可能需要您再检查一下。

jxbb233, 2024-01-04:
我尝试完全重新配置了一遍环境，但似乎问题还是存在......
我同学这两天会再跑一遍代码，看看这个问题能不能复现，issue就先开着吧

JunyuanDeng, 2024-01-05:
Yes, it's a bug and I fixed it now. Thanks for reminding! you may need to reinstall the third party.

closed 2024-01-13 by JunyuanDeng

peterWon, 2024-02-29:
@JunyuanDeng 作者你好，感谢分享代码。似乎上述问题还存在呢，我用的是新代码，subscene分支，跑的KITTI00