JunyuanDeng/NeRF-LOAM #21: AttributeError: 'NoneType' object has no attribute 'cuda'

hlldy6858, 2024-01-30:
你好，请问我这个问题是什么原因？ 显卡是3060 12G，torch1.10，CUDA11.1，但报错并不是因为爆显存

[PatchWorkpp::PatchWorkpp() - INITIALIZATION COMPLETE
Decoder(
  (pe): Same()
  (pts_linears): ModuleList(
    (0): Linear(in_features=16, out_features=256, bias=True)
    (1): Linear(in_features=256, out_features=256, bias=True)
  )
  (sdf_out): Linear(in_features=256, out_features=1, bias=True)
)
******* initializing first_frame: 0
initializing the first frame ...
PatchWorkpp::PatchWorkpp() - INITIALIZATION COMPLETE
******* tracking process started! *******
tracking frame:   0%|                                    | 0/99 [00:02<?, ?it/s]
Process Process-3:
Traceback (most recent call last):
  File "/home/hlldy/anaconda3/envs/nerf/lib/python3.8/multiprocessing/process.py", line 315, in _bootstrap
    self.run()
  File "/home/hlldy/anaconda3/envs/nerf/lib/python3.8/multiprocessing/process.py", line 108, in run
    self._target(*self._args, **self._kwargs)
  File "/home/hlldy/NeRF-LOAM/src/tracking.py", line 83, in spin
    self.do_tracking(share_data, current_frame, kf_buffer)
  File "/home/hlldy/NeRF-LOAM/src/tracking.py", line 101, in do_tracking
    decoder = share_data.decoder.cuda()
AttributeError: 'NoneType' object has no attribute 'cuda'
^C^CTraceback (most recent call last):
](url)

JunyuanDeng, 2024-01-30:
Hello,
Because we need the relative pose between t-1 and t-2 frames, which is inaccessible for first frame, so we do more tracking iteration for the first frame.

In your case, it seems that the tracking for the first frame is still running while the mapping need the tracked results.

So you may need wait more time for the first frame. You can adjust the  the code [here](https://github.com/JunyuanDeng/NeRF-LOAM/blob/2fe4e8d8dd9a8da12624ba8e8e4f49d851ce8c92/src/nerfloam.py#L45) to wait, e.g., 60s.

Hope this can solve your problem!

hlldy6858, 2024-01-30:
> Hello, Because we need the relative pose between t-1 and t-2 frames, which is inaccessible for first frame, so we do more tracking iteration for the first frame.
>
> In your case, it seems that the [...]

Thank you for your help!
I adjust 90s for the wait of code,but now there is the same problem,and the code is stuck at "insert keyframe" without moving.
Looking forward to your reply.

******* initializing first_frame: 0
initializing the first frame ...
PatchWorkpp::PatchWorkpp() - INITIALIZATION COMPLETE
******* tracking process started! *******
tracking frame:   0%|                                    | 0/99 [00:02<?, ?it/s]
Process Process-3:
Traceback (most recent call last):
  File "/home/hlldy/anaconda3/envs/nerf/lib/python3.8/multiprocessing/process.py", line 315, in _bootstrap
    self.run()
  File "/home/hlldy/anaconda3/envs/nerf/lib/python3.8/multiprocessing/process.py", line 108, in run
    self._target(*self._args, **self._kwargs)
  File "/home/hlldy/NeRF-LOAM/src/tracking.py", line 83, in spin
    self.do_tracking(share_data, current_frame, kf_buffer)
  File "/home/hlldy/NeRF-LOAM/src/tracking.py", line 101, in do_tracking
    decoder = share_data.decoder.cuda()
AttributeError: 'NoneType' object has no attribute 'cuda'
mapping process started!!!!!!!!!
frame id 1
trans  tensor([0., 0., 0.], device='cuda:0')
insert keyframe

JunyuanDeng, 2024-01-31:
Great! Happy to hear that!

JunyuanDeng, 2024-01-31:
By the way, you can uncomment all the `torch.cuda.empty_cache()` in files "src/variations/render_helpers.py" and "src/variations/voxel_helpers.py" to gain some efficiency. While this will centainly add some GPU memory.

closed 2024-01-31 by JunyuanDeng

hlldy6858, 2024-01-31:
> By the way, you can uncomment all the `torch.cuda.empty_cache()` in files "src/variations/render_helpers.py" and "src/variations/voxel_helpers.py" to gain some efficiency. While this will centainly add some GPU memory.

Thanks for your advice,if I have enough GPU memory,I will try to uncomment it. : )