JunyuanDeng/NeRF-LOAM #25: 关于patchwork++的问题

whwh747, 2024-04-02:
您好，非常感谢您出色的工作。我在部署该项目的时候出了一些问题。我已经安装好了patchwork++，运行演示代码也可以出现分割的结果，并且我替换了src/dataset/所有.py文件的patchwork_module_path ="/disk2/wh/3rd/patchwork-plusplus/python_wrapper"，（因为我并未在build中发现python_wrapper，python_wrapper在patchwork-plusplus中。）  我不知道patchwork++是应该装在base环境还是nerf-loam的虚拟环境中，我目前是装在了base环境中，因为我使用nerf-loam虚拟环境安装patchwork++时会出现X11不存在的问题，但我系统里确实安装了X11，我不知道会不会是这个问题，我使用的ubuntu22。希望您能帮助我解决这个问题。
(nerf_loam) wh@CFN-Titan-Server:~/code/NeRF-LOAM$ python demo/run.py configs/kitti/kitti_00.yaml
Traceback (most recent call last):
  File "demo/run.py", line 25, in <module>
    slam = nerfloam(args)
  File "/disk2/wh/code/NeRF-LOAM/src/nerfloam.py", line 31, in __init__
    self.data_stream = get_dataset(args)
  File "/disk2/wh/code/NeRF-LOAM/src/utils/import_util.py", line 5, in get_dataset
    Dataset = import_module("dataset."+args.dataset)
  File "/disk2/wh/anaconda3/envs/nerf_loam/lib/python3.8/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1014, in _gcd_import
  File "<frozen importlib._bootstrap>", line 991, in _find_and_load
  File "<frozen importlib._bootstrap>", line 975, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 671, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 843, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "/disk2/wh/code/NeRF-LOAM/src/dataset/kitti.py", line 12, in <module>
    import pypatchworkpp
ModuleNotFoundError: No module named 'pypatchworkpp'

JunyuanDeng, 2024-04-09:
Hi, Sorry for late reply!

can you provide more details and errors during the installation of patchwork++?

I encounter same problem before, normally, by commenting the line from [38](https://github.com/url-kaist/patchwork-plusplus/blob/dc33f599aae03b7f2388b429adc39d5403347ee6/CMakeLists.txt#L38) to [57](https://github.com/url-kaist/patchwork-plusplus/blob/dc33f599aae03b7f2388b429adc39d5403347ee6/CMakeLists.txt#L57), the problem can be solved.

whwh747, 2024-04-10:
thank you，it is work！！！I will close this issue @JunyuanDeng

closed 2024-04-10 by whwh747