JunyuanDeng/NeRF-LOAM #29: 八叉树代码的困惑

whwh747, 2024-06-21:
很抱歉再次打扰你，我对八叉树的代码有些困惑，希望你可以不吝赐教。
1.只在叶节点（树中节点的type==SURFACE）对应的体素上的八个顶点才有embedding吗？
2.八叉树中NONLEAF,SURFACE和FEATURE分别表示的是什么样的节点呢？
3.get_centres_and_children()函数中all_voxels中存放的是什么？每一个树节点的side_表示的是什么呢？
希望得到你的回复，感谢！

JunyuanDeng, 2024-06-24:
您好，我这边直接调用三方库直接获取voxel的顶点信息，具体的我没深入了解。
抱歉不能回答你的问题，我对他的理解，都是在这个[函数[(https://github.com/JunyuanDeng/NeRF-LOAM/blob/2fe4e8d8dd9a8da12624ba8e8e4f49d851ce8c92/src/mapping.py#L320)里面了，也就是map_states的key和get_centres_and_children结果的对应关系。

JunyuanDeng, 2024-06-24:
您好，我这边直接调用三方库直接获取voxel的顶点信息，具体的我没深入了解。
抱歉不能回答你的问题，我对他的理解，都是在这个[函数](https://github.com/JunyuanDeng/NeRF-LOAM/blob/2fe4e8d8dd9a8da12624ba8e8e4f49d851ce8c92/src/mapping.py#L320)里面了，也就是map_states的key和get_centres_and_children结果的对应关系。

whwh747, 2024-06-24:
感谢你的回复！我会关闭这个问题。祝好！

closed 2024-06-24 by whwh747