JunyuanDeng/NeRF-LOAM #31: Same SDF issue

woojinKwon, 2024-07-11:
Hello,

I am currently running demo/run.py with the maicity00 dataset, and I have encountered an issue. After initializing the first frame, when mapping the subsequent frames, all valid SDF values for the samples have the same value. This issue leads to a total_verts size of 0 during the mesh construction process. I want to highlight that this issue occurs consistently and is not intermittent. I have checked the sample points values and indices, and they seem to be normal.

Is there a solution to this issue?

JunyuanDeng, 2024-07-16:
Hello,

Is the mesh of first frame looks good?

woojinKwon, 2024-07-16:
Hello, thank you for your help.
What I previously mentioned as "After initializing the first frame" refers to the execution of self.initialized = True after the first operation of the self.do_mapping function within the spin() function of mapping.py. During the initial do_mapping process, an issue occurred where all SDF values had the same value. As a result, the first frame mesh was not formed correctly.

JunyuanDeng, 2024-07-18:
That's so weird.

Before the first do_mapping process, all voxel embedding is assigned with zero, it's after several do_mapping that the SDF (so does the voxel embedding) would be modified. But I have no idea why this occurs, we have not met this before...

closed 2024-11-06 by JunyuanDeng