JunyuanDeng/NeRF-LOAM #33: poor result about mesh

Zhangjyhhh, 2024-07-16:
I run your code use default config on new college dataset, but i can not get the same result as table 1 show. By the way, i use shine_mapping script to evaluate. Here are my result:

JunyuanDeng, 2024-07-16:
Thanks for your attention.

Is mesh that used to evaluate is the one after "key-scan refine"?

JunyuanDeng, 2024-07-16:
You could upload a screenshot instead of a mesh file so that I can provide more help.

Zhangjyhhh, 2024-07-16:
> Thanks for your attention.
> 
> Is mesh that used to evaluate is the one after "key-scan refine"?

I use the final_mesh.ply. Is the mesh that used to evaluate is the one after "key-scan refine
[image]

JunyuanDeng, 2024-07-16:
Yes, what is mesh looks like?

Zhangjyhhh, 2024-07-16:
> You could upload a screenshot instead of a mesh file so that I can provide more help.

here are my evaluation screenshot , config screenshoot and mesh screenshot
[image]
[image]
[image]
[image]

> Yes, what is mesh looks like?

here are my evaluation screenshot , config screenshoot and mesh screenshot
[image]
[image]
[image]
[image]

Zhangjyhhh, 2024-07-16:
> Yes, what is mesh looks like?

By the way, i only change the parameter " read_offset" from  default 5 to1, does it will change mesh quality

JunyuanDeng, 2024-07-16:
Yes, it would decrese the mesh quality, newer college dataset is collected by handhold devices, which means the lidar is not only suffer from motion distortion but also too dense for the task of reconstruction. Basically, the shape of the mesh is good, but the ground is not flat, that may due to the dense lidar scan and motion distortion. You can try to set "read_offset" to 1 to see the results.

Zhangjyhhh, 2024-07-16:
> Yes, it would decrese the mesh quality, newer college dataset is collected by handhold devices, which means the lidar is not only suffer from motion distortion but also too dense for the task of [...]

the above result is from read_offset=1, and I will try read_offset=5 later,and thanks for your reply

JunyuanDeng, 2024-07-16:
Hope you get better results~

Zhangjyhhh, 2024-07-17:
> Hope you get better results~

This time i get better result same as table 1:
{'MAE_accuracy (m)': 0.12597577918410047, 'MAE_completeness (m)': 0.17532213101945665, 'Chamfer_L1 (m)': 0.15064895510177856, 'Chamfer_L2 (m)': 0.2505973745191964, 'Precision [Accuracy] (%)': 77.41172923137316, 'Recall [Completeness] (%)': 71.95672924778981, 'F-score (%)': 74.58461977342017, 'Spacing (m)': 0.02, 'Inlier_threshold (m)': 0.2, 'Outlier_truncation_acc (m)': 0.4, 'Outlier_truncation_com (m)': 2.0}
Thanks for your quick reply!!!!!

closed 2024-07-17 by Zhangjyhhh

JunyuanDeng, 2024-07-18:
You even got better results than that in the paper, hahahah