JunyuanDeng/NeRF-LOAM #38: Bad result on hilti 2022 dataset

AbdelFarhan, 2025-01-20:
Hi,
 
I am trying your work on hilti 2022 dataset exp04 and I am getting very bad results.

Here is the reconstructed mesh
[image]

Here is the point cloud

[image]

Thanks in advance.

JunyuanDeng, 2025-01-20:
Hi,

May I ask which config file you used.

This sequence is men-held sequence from my search in google. So it's better set read_offset to be larger, like we set in configs/ncd/ncd_quad.yaml.

AbdelFarhan, 2025-01-20:
Thank you for your comment

here are the configs

`log_dir: './'
decoder: lidar
dataset: hilti

criteria:
  sdf_weight: 10000.0
  fs_weight: 1
  eiko_weight: 0.1
  sdf_truncation: 0.30

decoder_specs:
  depth: 2
  width: 256
  in_dim: 16
  skips: []
  embedder: none
  multires: 0

tracker_specs:
  N_rays: 2048
  learning_rate: 0.04
  step_size: 0.1
  max_voxel_hit: 20
  num_iterations: 30

mapper_specs:
  N_rays_each: 2048
  use_local_coord: False
  voxel_size: 0.4
  step_size: 0.2
  window_size: 5
  num_iterations: 15
  max_voxel_hit: 20
  final_iter: True
  mesh_res: 2
  learning_rate_emb: 0.002
  learning_rate_decorder: 0.005
  learning_rate_pose: 0.001
  freeze_frame: 20
  keyframe_gap: 8
  remove_back: False
  key_distance: 20

debug_args:
  verbose: False
  mesh_freq: 500
`
`base_config: hilti.yaml

exp_name: ./

data_specs:
  data_path: './ros/pcd'
  use_gt: False
  max_depth: 100
  min_depth: 1

tracker_specs:
  start_frame: 0
  end_frame: -1
  read_offset: 1

`

I am using  NVIDIA GeForce RTX 3090 24 GB. 
apparently it is not sufficient for my dataset. so I had to reduce some configurations in sparse voxel octree as follow:

https://github.com/JunyuanDeng/NeRF-LOAM/blob/2fe4e8d8dd9a8da12624ba8e8e4f49d851ce8c92/src/mapping.py#L85
I changed it to
self.svo.init(128*128*4, embed_dim, self.voxel_size)

https://github.com/JunyuanDeng/NeRF-LOAM/blob/2fe4e8d8dd9a8da12624ba8e8e4f49d851ce8c92/src/variations/voxel_helpers.py#L97
I changed it to
G = min(64, int(2 * 10 ** 9 / (points.numel() + children.numel())))

I will also try a larger set offset. But also, please let me know if my changes are going to reduce the accuracy significantly. If yes, then what other parameters I could change to fit my training in 24 GB

Thanks in advance

JunyuanDeng, 2025-01-20:
24 GB is sufficient, you met out of memory problem?

AbdelFarhan, 2025-01-20:
yes

AbdelFarhan, 2025-01-20:
Hi,

I am still getting poor results even after increasing the read_offset. Do you have any suggestion what could be the problem?